Marlinski/Rumble #1: Change Open Source to Free Software

PrasannaVenkadesh, 2015-12-03:
I hope, you would have already guessed what this issue is about from the title. You deserve my respect for developing this application given it's political use-case (not just Hong Kong case) but I see it along with a wider socio-economic aspect. That's how I see Free Software.

Applications like these that are Peer-to-Peer in nature which doesn't depend on any centralized infrastructure are most necessary to bring a radical change in how people produce and share information and that too when the knowledge of this is available for all to build upon which insists on freedom, it's unstoppable.

I very much appreciate the fact that you have chosen to use a Free Software, Copyleft GNU GPL for this program. But why not name it "Free Software License" instead of "Open Source License" in Settings?

Marlinski, 2015-12-04:
I am not sure what setting you are talking about, I add a LICENSE file GPL v3 in case there was any doubt about the license used.

Marlinski, 2015-12-04:
Just seen your pull request, you were talking about the Rumble's Setting :) 
I merged it.

closed 2015-12-04 by Marlinski

PrasannaVenkadesh, 2015-12-04:
That's Awesome man. Thanks for merging the PR :+1: