Marlinski/Rumble #3: Devices not discoverable under Android Wi-fi Hotspot

PrasannaVenkadesh, 2015-12-07:
I am running Android 4.1.2, peers are discoverable on Bluetooth and also on Infrastructure based Wi-Fi networks (like I am able to discover peers connected to my wireless router) but when I activate the Wi-Fi hotspot mode the application creates a open hotspot named "Rumble" and the peers can connect to that SSID, but then they are not discoverable each other.

I have tested this with more than 3 devices. I think this has to be given top priority, because the app intended to work on Infrastructure-less network.

Marlinski, 2015-12-08:
I noticed too that sometimes peers fail to discover each other with WiFi but that powering off and on the wifi interface solved the problem. This has to do with the MulticastLock but I have yet to pinpoint exactly the issue. Can you confirm that the peers fail to discover, even after a restart of wifi interface ?

PrasannaVenkadesh, 2015-12-08:
Today I tested again and found the following.
-----

**Scenario 1**

Number of Android Phones: 2
Android Version: >= 4.1.2
Initial Setting: Wifi Off, Bluetooth Off

Started Rumble with `Initial Setting` in both phones, 
 1. Switched on `Hotspot` in one phone from `Network Drawer`
 2. Connected to the Open `Rumble` Hotspot in another phone from Network Drawer of Rumble.

**Observation: Peers cannot see, send (or) receive anything.**

-----

**Scenario 2**

Laptop: 1
Number of Android Phones: 2
Android Version: >= 4.1.2
Initial Setting: Wifi off, Bluetooth Off

1. Hotspot from Laptop (Open without Security).
2. Started Rumble in both phones with `Initial Setting`
3. From `Network Drawer` switched on `Wi-fi` not hotspot in both phones.
4. Both phones connected to Open Wifi from Laptop.

**Observation: Peers can see, send and receive messages.**

----

**Scenario 3**

Number of Android Phones: 3
Android Version: >= 4.1.2
Initial Setting: Wifi off, Bluetooth Off

With `Initial Setting` on three phones

1. Started Rumble.
2. Hotspot via `Network Drawer` from only one Android Phone.
3. Switched Wi-Fi on the rest of the two Android Phones via `Network Drawer`.

**Observation:**

1. Hotspot phone cannot see any devices.
2. Other two devices can see each other, but not the hotspot phone.
3. Other two devices can send / receive data.
4. Updates from other two devices not reflected in hotspot phone (expected since it cannot discover it's peers) also the vice versa.

-----

**Scenario 4**

Same as **Scenario 3** with the following,

1. Switched on Bluetooth on the hotspot phone and one of the another phone.

**Observation**:

1. Observation 4 of Scenario 3 has been overcome (i.e All three devices can send or recieve 

Based on the above test cases and their respective observation, I came to a conclusion that, no matter how many devices participate, the one

Marlinski, 2015-12-08:
Thank you for taking the time to run those tests. You were right in your analysis, this was due to a failure from Android to join a multicast group in hotspot mode unless I specify explicitely which interface to use.

I just pushed a fix which solved the issue.

closed 2015-12-08 by Marlinski

PrasannaVenkadesh, 2015-12-08:
This is awesome and thanks for pushing a fix. I have found some bugs related to User Interfaces too. Will report them as soon as possible and by the way, I saw Rumble was submitted for the F-Droid Queue, is it accepted there or still in the review process?

Marlinski, 2015-12-09:
Still in review, there hasn't been any answer to my request yet (https://f-droid.org/forums/topic/rumble/), I will try to make the f-droid recipe myself.