Marlinski/Rumble #4: Unable to Find the Data Policy

PrasannaVenkadesh, 2015-12-10:
@Marlinski Under `Statistics` you have mentioned about data collection and a choice to opt out of the data collection. You have mentioned the URL as `http://disruptedsystems/rumble/datapolicy` which I suppose should be `http://disruptedsystems.org/rumble/datapolicy` instead.

Nevertheless, I am not able to find the document to get more information on the intention of the data collection and how it is being processed.

Marlinski, 2015-12-10:
The data policy is actually here:
http://disruptedsystems.org/documentation/#statistics

I will have this text updated !

Marlinski, 2015-12-14:
I have modified the link to point to the right URL.
https://github.com/Marlinski/Rumble/commit/a2bba93a1b3dd9413db9c2e4256c387769f0c3de

closed 2015-12-14 by Marlinski