Marlinski/Rumble #5: software buttons obscure app

pcvonz, 2015-12-14:
Hey, the software buttons on my m8 are drawn on top of your app.

Marlinski, 2015-12-14:
Do you use android lollipop by any chance ?

TheLastProject, 2015-12-14:
I see this issue on Android 5.1.1 too. Screenshot:
[image]

PrasannaVenkadesh, 2015-12-15:
@Marlinski remember, I commented about some UI related bugs, this is it. I am not sure if this is the problem with different android versions (or) different screen sizes. I am running Android 4.1.2 and the screensize is 4".

I have seen this bug on devices with screensize more than 4" and also interestingly those devices were also running Android 5, that is why I am not able confirm if it is the problem due to screensize or Android version.

taranarmo, 2015-12-15:
[image]
Android 6.0.1, Nexus 7 2nd generation.

PrasannaVenkadesh, 2015-12-15:
Ok, just now confirmed with another device. Issue is not due to screen size but because of Android versions 5 and above.

Marlinski, 2015-12-16:
I'll check this out thanks for the feedback.

viric, 2015-12-30:
As additional info, it does not only happens for the bottom soft buttons, but also for the top android indication bar (tested in android 6).

PrasannaVenkadesh, 2016-01-02:
@Marlinski any progress on this? I think we need to give this bug high priority as we encounter a handful of people encountering this bug and this leaves the app unusable at times. I am ready to fix it, if you have identified where the bug is.

xoir, 2016-01-05:
This screenshot might show a connected problem. In this case it is the taskbar that is overlaying the app screen an making it hard/impossible to add a group.

[image]

Dbof, 2016-01-21:
I am having this problem, too. Fixed it like this: https://github.com/Dbof/Rumble/commit/13c32db95aec297c57baf8c35a98a1faa709ffd2

Dbof, 2016-01-23:
Just noticed the same problem on the login screen, when selecting text. The toolbar is overlapped by the statusbar.

Dbof, 2016-01-23:
Found the problem: Seems to be an incompatibility between Material Design and SlidingMenu library. See this issue for more: https://github.com/jfeinstein10/SlidingMenu/issues/680
This solution by @Den-Rimus fixes the problem:

> The best idea is to fix this in SlidingMenu's code.
> 
> In file
> YOUR-PROJECT-UNDER-GRADLE/sliding-menu/src/com/jeremyfeinstein/slidingmenu/lib/SlidingMenu.java,
> protected boolean fitSystemWindows(Rect insets) method replace
> 
>         int bottomPadding = insets.bottom;
> 
> with next:
> 
>         int bottomPadding = insets.bottom;
>         if (Build.VERSION.SDK_INT >= 21) {
>             Resources resources = getContent().getResources();
>             int resourceId = resources.getIdentifier("navigation_bar_height", "dimen", "android");
>             if (resourceId > 0) {
>                 bottomPadding += resources.getDimensionPixelSize(resourceId);
>             }
>         }

SimJoSt, 2016-08-19:
Is there any noticeable progress on this?
It's the number on issue with recommending the app to other people. It just looks broken...

Dbof, 2016-08-19:
Unfortunately, @Marlinski seems to have abandoned this project. If you want to take over, I already provided a solution to the problem in my last comment.

Marlinski, 2016-08-19:
I did not abandon the project, I wish I have more time! I will try to pick up with the issues :)

SimJoSt, 2016-08-19:
Testing wise I can support you anytime. Coding wise, maybe in 2 years ;-)

SimJoSt, 2016-08-21:
I have no idea, why there was no pullrequest made, by there was a proposed solution for this problem https://github.com/Marlinski/Rumble/issues/5#issuecomment-174187774

Dbof, 2016-08-21:
@SimJoSt Because the problem is in the [SlidingMenu](https://github.com/jfeinstein10/SlidingMenu/) library and is does not seem to be regularly updated. You need to fork it yourself and change the code.

Marlinski, 2016-08-21:
So I should fork the project and fix it myself, maybe it would be best if it was fixed upstream since SlidingMenu is a git submodule in Rumble (thanks to you dbof).

Marlinski, 2016-08-21:
I followed @Dbof  advice, can anyone test wether this patch worked ?

SimJoSt, 2016-08-21:
@Marlinski I tried setting up Android Studio and Gradle but failed.
If you give me an APK or a manual for the setup then I will test it for you.

Marlinski, 2016-08-22:
@SimJoSt It was a pain but I finally set the git submodule right. If you just clone recursively the project it should compile without issue:

git clone --recursive -j8 git@github.com:Marlinski/Rumble.git

I don't want to provide development apk as I prefer people to stick with either f-droid or google play so that I can push update :)

SimJoSt, 2016-08-22:
@Marlinski I will try to compile it again as soon as I can, but I didn't have any problems with the submodule. Android studio didn't accept Gradle 2.2, which is required for your project. It only allows for 2.14.1 and newer...

Marlinski, 2016-08-22:
@SimJoSt I have added gradle wrapper files that were missing. you can try now.

SimJoSt, 2016-08-22:
@Marlinski Perfect. The problem is gone.
Still, I get an error:
> Gradle 'Rumble' project refesh failed
> Error: Configuration with name 'default' not found.

Marlinski, 2016-08-22:
@SimJoSt you are talking about the UI problem which is gone ? should I close the issue ?
If you have an issue with the build please open another ticker :-)

SimJoSt, 2016-08-22:
Sorry for not being clear about hits :-)
The Gradle Version error was gone. The wrapper you added is working.
But I still have problems to import it properly into Android Studio (#45)

Marlinski, 2016-08-23:
Issue is fixed.

closed 2016-08-23 by Marlinski

SimJoSt, 2016-08-23:
I concur. The problem is fixed.