Marlinski/Rumble #6: Hardware compatibility in f-droid

taranarmo, 2015-12-15:
Application requires camera, I suppose, which my tablet (Nexus 7, 2nd gen) has, but  f-droid indicates that app is not compatible with.
BTW application works fine.

Marlinski, 2015-12-16:
The camera is used to scan a group invitation (in the form of a QRCode).
I suppose I can safely remove this requirement, I'll do some check, thanks for letting me know.

PrasannaVenkadesh, 2015-12-16:
I installed Rumble through Play Store, the app lists my device as not compatible in f-droid. I am not able to find which permission makes my device incompatible in f-droid.

taranarmo, 2015-12-31:
Today I've got upgrade from f-droid and saw that rumble requires camera flash, not only camera. So, I think it is safe to remove that requirement.

closed 2016-08-21 by Marlinski