Marlinski/Rumble #17: login button on the inital screen is confusing

xoir, 2015-12-29:
Actually you are not providing a login by the meaning of the word "Login". To not confuse privacy-orientated people I would recommend to rename the button to "Start". I think this would make much more sense there.

The screen is the first thing people will see. Let's make sure we are not scaring them away.

viric, 2015-12-30:
Additionally, for me the login button was completely hidden by the Android softkeyboard, and "Enter" added a newline in the login. Hiding the keyboard allowed me to see and press Login.

Marlinski, 2015-12-30:
I close this issue, will solve the keyboard problem in a future fix

closed 2015-12-30 by Marlinski

viric, 2015-12-30:
For your information, this sounds to me like a very weird reason to close an issue: a future commit. That will break the link between the issue and the solution, and may end up with people creating duplicates of this.

viric, 2015-12-30:
Ah sorry. Now I read better that this was only about the login text. I should have created a new issue for the keyboard problem.