Marlinski/Rumble #20: enable the users to change the Bluetooth visibility time to their needs

xoir, 2015-12-29:
Whenever I switch my phone (Android 4.4.2) to the third state I am asked to allow that for 3600 seconds (1h).

Is there a way to increase that time? Is there a way to prevent auto-disabling?

Dbof, 2016-01-21:
According to this issue (https://code.google.com/p/android/issues/detail?id=6348), it is not really possible to exceed the 3600 seconds for discovery. There are a few workarounds, but you probably need root to solve this.

Marlinski, 2016-08-21:
as @dbof said, It is unfortunately not possible and I even heard that on certain phone there is a hard limit of 5 minutes :( I don't know yet how to make bluetooth discoverability indefinite as it is not supported officially by android. I close the comment as there is not much I can do about it.

closed 2016-08-21 by Marlinski