Marlinski/Rumble #22: sort statuses by incomming time vs sort them by sending date

xoir, 2015-12-29:
I am not sure which of both orders are to prefer that's why I would like to see both implemented. Right now all microblog-style status messages are sorted by incoming date.

Maybe for shaping the view it would make sense to keep the sorting in the status-tab as is and switch to "sorted by sending date" in the group-view-panel (while reading the content of a specific group (e.g. rumble.public)).

Dbof, 2016-01-21:
Problem with sending date is: It is strange behavior to put incoming messages in between two other messages somewhere in the list. Imagine getting many messages at once, and the message you are looking for gets inserted at the bottom of the list. You probably would not see it immediately and would need to scroll up and down to find new messages.
Incoming date is determined when a message is received, so it is logical to show it then.

Marlinski, 2016-08-21:
Yes you are both right and I wasn't exactly sure either what was the "right" way to sort the messages. Another sorting would be unread on top. Actually I already have all the mechanics inside that allows different sorting so it is just a matter of adding few button. Stay tuned !