Marlinski/Rumble #28: Display list of connected nodes

deavmi, 2016-01-14:
Display list of connected nodes

PrasannaVenkadesh, 2016-01-15:
Hi @deavmi can you be more descriptive about this?

 - If you are looking for the list of connected peers around you, it's already there on the Network Drawer (Swipe in left direction to reveal).
 - If you are looking for list of all users connected with you through rumble (even those whom you have not met but when you both belongs to same group) that is available under "Contact" section from Navigation Drawers (swipe right to reveal).

If you are asking about something else other than these, please describe it in detail

deavmi, 2016-01-15:
Oh okay, never mind then.

closed 2016-01-15 by deavmi

deavmi, 2016-01-15:
And a global peers list. Sounds like a bad idea though. Will cause spamming.