Marlinski/Rumble #29: Private Chats

PrasannaVenkadesh, 2016-01-16:
Allow peers around to chat securely & privately apart from chatting in a public channel.

What I would expect,

- Share text.
- Share pictures / files.
- Encrypted.

deavmi, 2016-01-16:
I'd like this. Also I think all data should be encrypted.

deavmi, 2016-01-16:
Maybe a file sharing feature that shows a lost of available files on the network could be nice.

Dbof, 2016-01-21:
While there is already the possibility to chat privately by creating a group, I support the request to create a real (opportunistic) one-to-one communication. I think using a Public-Key encryption and exchanging the public keys using the QR-codes, as already done with groups, would be a good solution.

deavmi, 2016-01-22:
Physical key exchange is good.

Marlinski, 2016-08-27:
This is an interesting use case though the obvious issue is that a message may never arrive to the intended destination. If I were to implement this, it would requires that other users should altruisticly host encrypted file that they could not themselves read. Since this content is not viewable by the carrier (unless he is the intented destination), it can be abused by spammers. But I will dig into this it is interesting, will try to find a reasonable tradeoff between epidemic routing, and altruistic hosting in the settings.

deavmi, 2016-08-31:
Ok. cooleo.