Marlinski/Rumble #31: WiFi on android sucks

deavmi, 2016-01-16:
Maybe a message should be displayed when using wifi mode about how the infrastructure mode is not the best. With an option to "Not show this message again.". I think this is important as infrastructure mode is crap.

Marlinski, 2016-08-21:
can you develop as to why infrastructure mode is crap ?

deavmi, 2016-08-21:
It's all in the word.

Ad-hoc mode allows the network to operate in a more democratic way. If 
it uses infrustructure mode then if the AP goes down then all devices 
connected to it cannot communicate to each other using WiFi as the medium.

Marlinski, 2016-08-22:
ah well, I couldn't agree more unfortunetly google has been deaf to the issue since 2008:
https://code.google.com/p/android/issues/detail?id=82

Even though wifi chips do provide adhoc support, it is currently not natively supported. The only way would be to root the device (as done on the serval project) which is not a requirement I want.

closed 2016-08-22 by Marlinski

deavmi, 2016-08-22:
Google should just implement it. They are such bullshitters sometimes, honestly.