Marlinski/Rumble #34: WiFi should be used differently

deavmi, 2016-01-16:
Since Bluetooth allows for a true ad-hoc mode. Why not use Bluetooth for the mesh and use wifi to broadcast a hotspot for people to connect to with the SSID (offline social network) which once connected android will ask them to sign into the wifi and make them land on a captive portable describing a little bit about rumble with a locally hosted rumble.apk file. :)

deavmi, 2016-01-16:
This should be optional though.

Marlinski, 2016-01-22:
The captive portal is a good idea, unfortunately it requires root in order to set up the appropriate iptable rule (you want every request made by clients to be redirected to your local address) I have yet to found a way to do captive portal without root

deavmi, 2016-01-23:
Surely you can run a captive portal using a web server on android and then binding it to the router (phone's  thering) IP address at port `80` hence by default if the user visits the router's page he can get a captive portal.