Marlinski/Rumble #35: Internet connecting many Rumble networks

deavmi, 2016-01-16:
There should be a way to get Rumble to work over the net using Tor.

Dbof, 2016-01-21:
In my opinion...no, there should not. There is a specific use case for Rumble, and it is clearly an offline network. Making it connect online only adds complexity and reduces the overall security of the networl, as they are routed through Tor, which comes with more complications regarding privacy and security.

PrasannaVenkadesh, 2016-01-21:
Agreed with @Dbof. Rumble is "Off the Grid".

bcu8heq2, 2016-03-26:
Agree.

deavmi, 2016-03-27:
Yeah. This is a good idea. But remember to encrypt the internet communications as well.

Marlinski, 2016-08-21:
I believe this is already a feature implemented in the briar project. Though I am seduced by the Idea of using TOR and hidden services as a way to bridge people,  Rumble does not intend to work over the Internet at the time.

closed 2016-08-21 by Marlinski