Marlinski/Rumble #36: Progress Bar While Scanning in WiFi Mode

PrasannaVenkadesh, 2016-01-21:
@Marlinski I am trying to mimic the progress bar that appears between `BluetoothScanStarted` and `BluetoothScanEnded` event for Wifi Scanning.

I can see the `WifiScanner.java` class, and also I see this particular log statement,

    Log.d(TAG, "((( Wifi Scanner started )))");

but this statement is not logged when I checked with

    adb logcat WifiScanner:D *:S

BluetoothScanner log statements are getting logged on Scanner Started, intervals between scanning and when it ends.

Marlinski, 2016-08-21:
If you are interested in the Bluetooth Scanner you might want to look at the BluetoothScanner.java file instead of WifiScanner.java. 

Regarding wifi connections, it is all managed by the android system, I don't manage them myself I just listen at the BroadcastIntent throws by android such as CONNECTED of DISCONNECTED. The WifiScanner.java is not currently used by the project. My goal would be to automatically connect to rumble access point only if the user isn't connected to an AP already but I haven't done it yet.

Marlinski, 2016-08-21:
there is no progress bar in wifi because unlike bluetooth, scanning is done constantly in WiFi (passive scanning by listening to beacon packet sent from nearby AP), though active scanning can be done as well, it doesn't disrupt the connections quiet as much as in bluetooth.

closed 2016-08-21 by Marlinski