Marlinski/Rumble #44: photo is only send when text is added

SimJoSt, 2016-08-20:
If you try to post a photo, it will not be send and no error message is displayed.
Only if additionally text is added to the post, the photo will be send.

Marlinski, 2016-08-21:
would you prefer I add notification or should you prefer that it would be possible to send a picture without texte ?

SimJoSt, 2016-08-21:
Good question.
I believe allowing to post photos without text is the more preferable option. I can't think of a reason to disallow it.

Marlinski, 2016-08-21:
Since every status is passed around, you cannot easily trace back the context/origin of a photo. I guess I thought it would be better to force some text, but I agree the implementation was lacking at least a notification :) I'll fix it

closed 2016-08-21 by Marlinski

SimJoSt, 2016-08-21:
Perfect :)
But just so I understand: How is a photo post any different from a text post regarding it's difficult to trace origin?