Marlinski/Rumble #46: build error: :app:dexDebug

SimJoSt, 2016-08-22:
I've got the following error, while trying to build:
https://gist.github.com/SimJoSt/672f8bee7ed27f2ed9b78ffe4a3ecaa1

It appears to be a problem with duplicate libraries (different versions) or with too many libraries overall.

Marlinski, 2016-08-22:
which version of android-studio are you running ? It's been a while since I upgraded mine :p 

You can take a look at this issue discussion here and update with the results:

http://stackoverflow.com/questions/20989317/multiple-dex-files-define-landroid-support-v4-accessibilityservice-accessibility

Marlinski, 2016-08-23:
I just upgraded android-studio which advised me to upgrade gradle, so I did.
Can you try now ? (start from a fresh install, remove everything and git clone the project).

SimJoSt, 2016-08-23:
I did. No problems at all.
Issue is resolved.

closed 2016-08-23 by SimJoSt