Marlinski/Rumble #48: Cannot checkout in SourceTree

virl, 2016-08-26:
```
Cloning into 'libs/SlidingMenu'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:Marlinski/SlidingMenu.git' into submodule path 'libs/SlidingMenu' failed
Cloning into 'libs/SlidingMenu'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:Marlinski/SlidingMenu.git' into submodule path 'libs/SlidingMenu' failed
Cloning into 'libs/SlidingMenu'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:Marlinski/SlidingMenu.git' into submodule path 'libs/SlidingMenu' failed
Cloning into 'libs/SlidingMenu'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:Marlinski/SlidingMenu.git' into submodule path 'libs/SlidingMenu' failed
Completed with errors, see aboveCompleted with errors, see above
```

Marlinski, 2016-08-26:
I should replace the git@github.com with https://github.com, thanks for notifying me.

virl, 2016-08-26:
@Marlinski By the way, I'm working on something similar (as I shown you previously): http://underdark.io

Want to discuss some ideas about P2P messaging?

closed 2016-08-27 by Marlinski

Marlinski, 2016-08-27:
you can contact me using email or on IRC (freenode) channel #disruptedsystems

virl, 2016-08-27:
@Marlinski Ok, wrote you email.