Marlinski/Rumble #51: Use Internet if available

davidak, 2016-09-07:
would this be in the scope of this project?

A lot of people have internet on their mobile devises, so if this app supports it, it would have a lot more users. That may lead to more contribution and a better app, also more testing.

If the app get's more stable this way, it's more helpful for people that have no internet or censored internet!

SimJoSt, 2016-09-18:
As the caption of the app is "OFF-THE-GRID COMMUNICATION" and in my opinion the Internet is "the grid", I never would expect the to have this functionality. I also wouldn't want to rule it out, but I believe it defeats the purpose of geographically close peer-to-peer communication.

davidak, 2016-09-20:
OK

closed 2016-09-20 by davidak