Marlinski/Rumble #59: Is Rumble compatible with SSB (Secure Scuttlebutt) protocol?

strypey, 2018-09-08:
Rumble seems to have similar project goals to the [Scuttlebutt network](http://scuttlebutt.nz/), which runs on the SSB (Secure Scuttlebutt) protocol. Is Rumble compatible with SSB apps, and would it be possible to bridge users of the two networks?

Marlinski, 2019-03-11:
It looks similar but Rumble doesn't run using the internet infrastructure to operate (though it could use internet link at times). All of the message exchanges are done when carriers of the mobile device (humans, though it could be also a special device) actually meet and the phones being within range of each other can connect. It is not compatible with SSB but I imagine it would be possible to build a bridge whenever Internet is available.

strypey, 2019-03-11:
This is exactly how SSB works too. From the [Scuttlebutt Protocol Guide](https://ssbc.github.io/scuttlebutt-protocol-guide/):
"Scuttlebutt is a protocol for building decentralized applications that work well offline and that no one person can control."

There are a number of working SSB  clients in use, including Patchwork for desktops, and a mobile client called [Manyverse](https://www.manyver.se/) that has a beta available for Android.

The [Serval Project](https://github.com/servalproject) is another decentralized comms project that aims to build device-to-device communication, without a hard dependence on a network connection.

I mention all of this not because I expect all decentralized tech projects to fold into one mega-project, I don't. But I do observe that it really helps boost adoption of new comms apps, if their users can connect not only to other users on the same app, but also users of other similar comms apps (shared network effect). I also think it's valuable for groups working on similar projects to be aware of each other so they can clearly communicate their points of difference to users, and maybe even learn from each other's experiments (both successes and failures). Thanks for your work on Rumble :)

chorliya, 2020-04-14:
> It looks similar but [..]

_looks_ dead as a doorknob

_.o0_